f00b4r0/uspot #4: uhttpd ucode error

AMArefkhani, 2024-02-12:
Hello. My configuration for uhttpd is as below :
config uhttpd 'main'
	list listen_https '0.0.0.0:443'
	list listen_https '[::]:443'
	option redirect_https '0'
	option home '/www'
	option rfc1918_filter '1'
	option max_requests '3'
	option max_connections '100'
	option cert '/etc/uhttpd.crt'
	option key '/etc/uhttpd.key'
	option cgi_prefix '/cgi-bin'
	list lua_prefix '/cgi-bin/luci=/usr/lib/lua/luci/sgi/uhttpd.lua'
	option script_timeout '60'
	option network_timeout '30'
	option http_keepalive '20'
	option tcp_keepalive '1'
	option ubus_prefix '/ubus'
	list listen_http '192.168.3.1:80'

config cert 'defaults'
	option days '730'
	option key_type 'ec'
	option bits '2048'
	option ec_curve 'P-256'
	option country 'ZZ'
	option state 'Somewhere'
	option location 'Unknown'
	option commonname 'OpenWrt'

config uhttpd 'uspot'
	list listen_http '10.0.0.1:80'
	option redirect_https '0'
	option max_requests '5'
	option no_dirlists '1'
	option home '/www-uspot'
	list ucode_prefix '/hotspot=/usr/share/uspot/handler.uc'
	list ucode_prefix '/cpd=/usr/share/uspot/handler-cpd.uc'
	option error_page '/cpd'
#	# if using TLS and/or supporting RFC8908 CapPort API:
#	#list listen_https '10.0.0.1:443'
#	option cert '/usr/share/certs/captive.pem'	# to be provided manually
#	option key '/usr/share/certs/captive.key'	# to be provided manually
#	# for RFC8908 support:
#	list ucode_prefix '/api=/usr/share/uspot/handler-api.uc'

# if using RADIUS UAM authentication:
#config uhttpd 'uam3990'
#	list listen_http '10.0.0.1:3990'
#	option redirect_https '0'
#	option max_requests '5'
#	option no_dirlists '1'
#	option home '/www-uspot'
#	list ucode_prefix '/logon=/usr/share/uspot/handler-uam.uc'
#	list ucode_prefix '/logoff=/usr/share/uspot/handler-uam.uc'
#	list ucode_prefix '/logout=/usr/share/uspot/handler-uam.uc'
#
when I start uhttpd service, it returns "daemon.err uhttpd[6084]: Error: Unable to open ucode handler: No such file or directory".

f00b4r0, 2024-02-13:
Hi, can you provide the output of the following commands:
```
opkg list-installed | grep ucode
opkg list-installed | grep uspot
```

f00b4r0, 2024-02-13:
uspot looks correct but you have multiple conflicting ucode module versions installed (2023-06-06 vs 2023-11-07) and two versions of libucode. Everything should be on 2023-11-07. Can you try to opkg update / opkg upgrade?

I suspect this is the cause of your problem.

This is what it should look like on e.g. 23.05.2:

```
liblucihttp-ucode - 2023-03-15-9b5b683f-1
libucode20230711 - 2023-11-07-a6e75e02-1
rpcd-mod-ucode - 2023-07-01-c07ab2f9-1
ucode - 2023-11-07-a6e75e02-1
ucode-mod-fs - 2023-11-07-a6e75e02-1
ucode-mod-html - 1
ucode-mod-log - 2023-11-07-a6e75e02-1
ucode-mod-math - 2023-11-07-a6e75e02-1
ucode-mod-nl80211 - 2023-11-07-a6e75e02-1
ucode-mod-rtnl - 2023-11-07-a6e75e02-1
ucode-mod-ubus - 2023-11-07-a6e75e02-1
ucode-mod-uci - 2023-11-07-a6e75e02-1
ucode-mod-uloop - 2023-11-07-a6e75e02-1
uhttpd-mod-ucode - 2023-06-25-34a8a74d-1
```

AMArefkhani, 2024-02-13:
and also here are the other configurations for uspot:
/etc/config/uspot:
 ``` 
#for auth mode 'credentials', add any number of the following config entry
#config credentials
#	option uspot 'example'
#	option username 'myuser'
#	option password 'mypass'

## Values provided for the options below reflect the defaults used when the option is not set.

config uspot 'captive'
	option auth_mode 'click-to-continue'		# one of 'uam', 'radius', 'credentials', 'click-to-continue'
	option idle_timeout '600'	# client is kicked when idle for more than N seconds, defaults to 600, option used if not provided by radius
	option session_timeout '1000'	# client is kicked if connected for more than N seconds, defaults to 0, option used if not provided by radius
	option interface 'captive'		# network interface (from config/network) on which captive clients will be managed#
	option setname 'uspot'		# firewall ipset name for client management
	option debug '0'		# turn on debugging output in logs

# captive portal API (RFC8908) configuration:
#	option cpa_can_extend '0'	# 'can-extend-session' is true if this option is set to '1', false otherwise
#	option cpa_venue_url ''		# value is provided verbatim as 'venue-info-url'

# for auth mode 'uam' and 'radius':
#	option auth_server ''		# radius authentication server name or address
#	option auth_port '1812'		# radius authentication server port
#	option auth_secret ''		# radius authentication server password
#	option auth_proxy ''		# radius authentication server proxy
#	option acct_server ''		# radius accounting server name or address
#	option acct_port '1813'		# radius accounting server port
#	option acct_secret ''		# radius accounting server password
#	option acct_proxy ''		# radius accounting server proxy
#	option acct_interval ''		# radius accounting interim interval override
#	option das_secret ''		# radius DAS secret
#	option das_port '3799'		# radius DAS listen port
#	option nasid ''			# radius NAS-Identitifer, UAM '&nasid='
#	option nasmac ''		# radius Called-Station, UAM '&called='
#	option mac_format ''		# MAC format specifier: 'aabbccddeeff', 'aa-bb-cc-dd-ee-ff', 'aa:bb:cc:dd:ee:ff' or the equivalent uppercase
#	option location_name ''		# radius WISPr-Location-Name

# for auth_mode 'uam':
#	option uam_port '3990'		# local UAM server port
#	option uam_secret ''		# remote UAM server password
#	option uam_server ''		# remote UAM server base url, e.g. "https://server.example.com/" - NB: trailing slash
#	option challenge ''		# UAM CHAP shared challenge
#	option final_redirect_url ''	# URL the client will be redirected to upon login. Special value 'uam' enables UAM 'success/reject/logoff' redirections URLs.
#	option mac_auth '0'		# Attempt MAC-authentication first
#	option mac_password ''		# Password sent for MAC-auth, defaults to MAC address
#	option mac_suffix ''		# Optional suffix appended to username for MAC-auth
#	option uam_sslurl ''		# optional base url to local UAM SSL (requires valid SSL setup in uhttpd UAM config), e.g. "https://uspot.lan:3991/" - NB: trailing slash
 ``` 
/etc/config/network:
 ``` 
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd2d:2536:6255::/48'
	option packet_steering '1'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'

config interface 'lan'
[... 322 lines cut ...]
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'

config dhcp 'captive'
	option interface 'captive'
	option start '2'
	option limit '1000'
	option leasetime '2h'
	# add the following for RFC8910 Captive Portal API - DNS name is setup below
	#list dhcp_option '114,https://captive.example.org/api'
	# optionally provide NTP server (if enabled on the device) - recommended for SSL cert validation
	list dhcp_option_force '42,10.0.0.1'

# add a local domain name for HTTPS support, name must match TLS certificate
config domain
	option name 'captive.example.org'
	option ip '10.0.0.1'

# if using optional dynamic hosts whitelist
config ipset
	list name 'wlist'	# match value with whitelist ipset name in config/firewall
	list domain 'my.whitelist1.domain'
	list domain 'my.whitelist2.domain'

 ```

AMArefkhani, 2024-02-14:
Hi, The reported error solved by upgrading Openwrt to 23.05.2 while the previous version was 23.05.0. But I'm wondering does the uspot captive portal redirect any unauthenticated client traffic to the login page or does the client have to go to the login page itself?

f00b4r0, 2024-02-14:
> But I'm wondering does the uspot captive portal redirect any unauthenticated client traffic to the login page or does the client have to go to the login page itself?

Unauthenticated **HTTP** traffic will be redirected. **HTTPS** however will not (it can't be), but most client device perform so-called "Captive Portal Detection" (CPD) on HTTP for that very reason.

Enabling the Captive Portal API provides a smoother user experience.

AMArefkhani, 2024-02-14:
Many thanks for your help. Could you please tell me how to configure uspot to connect to the freeradius server. I also have another question, does uspot implement accounting for freeradius? I mean daily (or weekly, etc.) usage and rate limit.

f00b4r0, 2024-02-14:
For RADIUS configuration see this section: https://github.com/f00b4r0/uspot/blob/53b8cb88a94a21ab2a5c74122ee9e4a9f1ad4c9b/files/etc/config/uspot#L21-L36

you will need at least `auth_server` and `auth_secret`.

uspot currently only implements session time accounting. Traffic accounting is on the TODO list (see end of README), it's coming hopefully soon.

AMArefkhani, 2024-02-15:
Thanks. I have problem with Radius authentication mode. The freeradius server is located in the wan side with ip address 192.168.205.161. When clients try to connect with username and password, the following error is shown in the logread.
 
Thu Feb 15 14:35:22 2024 user.err : radcli: rc_read_dictionary: rc_read_dictionary couldn't open dictionary /etc/radcli/dictionary: No such file or directory

The configuration for uspot, firewall and uhttpd is as below:
uspot:
```
config credentials
	option uspot 'captive'
	option username 'amirmohammad'
	option password 'aref'

## Values provided for the options below reflect the defaults used when the option is not set.

config uspot 'captive'
	option auth_mode 'radius'		# one of 'uam', 'radius', 'credentials', 'click-to-continue'
	option idle_timeout '600'	# client is kicked when idle for more than N seconds, defaults to 600, option used if not provided by radius
	option session_timeout '240'	# client is kicked if connected for more than N seconds, defaults to 0, option used if not provided by radius
	option interface 'captive'		# network interface (from config/network) on which captive clients will be managed
	option setname 'uspot'		# firewall ipset name for client management
	option debug '0'		# turn on debugging output in logs

# captive portal API (RFC8908) configuration:
	option cpa_can_extend '0'	# 'can-extend-session' is true if this option is set to '1', false otherwise
	option cpa_venue_url ''		# value is provided verbatim as 'venue-info-url'

# for auth mode 'uam' and 'radius':
	option auth_server '192.168.205.161'		# radius authentication server name or address
	option auth_port '1812'		# radius authentication server port
	option auth_secret 'xiaomi-router'		# radius authentication server password
#	option auth_proxy ''		# radius authentication server proxy
#	option acct_server ''		# radius accounting server name or address
#	option acct_port '1813'		# radius accounting server port
#	option acct_secret ''		# radius accounting server password
#	option acct_proxy ''		# radius accounting server proxy
#	option acct_interval ''		# radius accounting interim interval override
#	option das_secret ''		# radius DAS secret
#	option das_port '3799'		# radius DAS listen port
#	option nasid ''			# radius NAS-Identitifer, UAM '&nasid='
#	option nasmac ''		# radius Called-Station, UAM '&called='
#	option mac_format ''		# MAC format specifier: 'aabbccddeeff', 'aa-bb-cc-dd-ee-ff', 'aa:bb:cc:dd:ee:ff' or the equivalent uppercase
#	option location_name ''		# radius WISPr-Location-Name

# for auth_mode 'uam':
#	option uam_port '3990'		# local UAM server port
#	option uam_secret ''		# remote UAM server password
#	option uam_server ''		# remote UAM server base url, e.g. "https://server.example.com/" - NB: trailing slash
#	option challenge ''		# UAM CHAP shared challenge
#	option final_redirect_url ''	# URL the client will be redirected to upon login. Special value 'uam' enables UAM 'success/reject/logoff' redirections URLs.
#	option mac_auth '0'		# Attempt MAC-authentication first
#	option mac_password ''		# Password sent for MAC-auth, defaults to MAC address
#	option mac_suffix ''		# Optional suffix appended to username for MAC-auth
#	option uam_sslurl ''		# optional base url to local UAM SSL (requires valid SSL setup in uhttpd UAM config), e.g. "https://uspot.lan:3991/" - NB: trailing slash
```
firewall:
```
config zone
	option name 'captive'
	list network 'captive'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'

config redirect
	option name 'Redirect-unauth-captive-CPD'
	option src 'captive'
	option src_dport '80'
	option proto 'tcp'
	option target 'DNAT'
	option reflection '0'
[... 84 lines cut ...]
	# for RFC8908 support:
	list ucode_prefix '/api=/usr/share/uspot/handler-api.uc'

# if using RADIUS UAM authentication:
config uhttpd 'uam3990'
	list listen_http '10.0.0.1:3990'
	option redirect_https '0'
	option max_requests '5'
	option no_dirlists '1'
	option home '/www-uspot'
	list ucode_prefix '/logon=/usr/share/uspot/handler-uam.uc'
	list ucode_prefix '/logoff=/usr/share/uspot/handler-uam.uc'
	list ucode_prefix '/logout=/usr/share/uspot/handler-uam.uc'
```

The information of nas in the database of freeradius is as below:
```
+----+-----------------+-----------+------+-------+---------------+--------+-----------+-------------+
| id | nasname         | shortname | type | ports | secret        | server | community | description |
+----+-----------------+-----------+------+-------+---------------+--------+-----------+-------------+
|  2 | 192.168.3.1     | NULL      | NULL |  NULL | xiaomi-router | NULL   | NULL      | NULL        |
|  3 | 10.0.0.1        | NULL      | NULL |  NULL | xiaomi-router | NULL   | NULL      | NULL        |
|  4 | 192.168.205.202 | NULL      | NULL |  NULL | xiaomi-router | NULL   | NULL      | NULL        |
+----+-----------------+-----------+------+-------+---------------+--------+-----------+-------------+
```

f00b4r0, 2024-02-15:
Please don't use this closed issue to ask unrelated support questions.

> Thanks. I have problem with Radius authentication mode. The freeradius server is located in the wan side with ip address 192.168.205.161. When clients try to connect with username and password, the [...]

You need to provide your RADIUS dictionary files to libradcli (`/etc/radcli/dictionary` as indicated by the error message above), by default none are provided by the libradcli package.
Dictionary files are available from e.g. https://github.com/radcli/radcli/tree/master/etc

AMArefkhani, 2024-02-16:
I added the dictionary file provided in this link [https://github.com/radcli/radcli/blob/master/etc/dictionary](https://github.com/radcli/radcli/blob/master/etc/dictionary) to the /etc/radcli/ and all attributes used in the /usr/share/uspot/uspot.uc were inculeded in this file. When I try to login no error shown in the logread and also it returns invalid credentials in the UI. Also, When I check with Tcpdump and Wirshark in the Openwrt and Freeradius server respectively, no request is sent. However when I user radtest in Openwrt, the Freeradius server works correctly. Did I missed somethings in the below configurations?
Thank you so much.
uspot:
 ``` 
config credentials
	option uspot 'captive'
	option username 'amir'
	option password 'aref'

## Values provided for the options below reflect the defaults used when the option is not set.

config uspot 'captive'
	option auth_mode 'radius'		# one of 'uam', 'radius', 'credentials', 'click-to-continue'
	option idle_timeout '600'	# client is kicked when idle for more than N seconds, defaults to 600, option used if not provided by radius
	option session_timeout '60'	# client is kicked if connected for more than N seconds, defaults to 0, option used if not provided by radius
	option interface 'captive'		# network interface (from config/network) on which captive clients will be managed
	option setname 'uspot'		# firewall ipset name for client management
	option debug '0'		# turn on debugging output in logs

# captive portal API (RFC8908) configuration:
	option cpa_can_extend '0'	# 'can-extend-session' is true if this option is set to '1', false otherwise
	option cpa_venue_url 'mci.captive.portal'		# value is provided verbatim as 'venue-info-url'

# for auth mode 'uam' and 'radius':
	option auth_server '192.168.205.161'		# radius authentication server name or address
	option auth_port '1812'		# radius authentication server port
	option auth_secret 'xiaomi-router'		# radius authentication server password
#	option auth_proxy ''		# radius authentication server proxy
#	option acct_server ''		# radius accounting server name or address
#	option acct_port '1813'		# radius accounting server port
#	option acct_secret ''		# radius accounting server password
#	option acct_proxy ''		# radius accounting server proxy
#	option acct_interval ''		# radius accounting interim interval override
#	option das_secret ''		# radius DAS secret
#	option das_port '3799'		# radius DAS listen port
#	option nasid ''			# radius NAS-Identitifer, UAM '&nasid='
#	option nasmac ''		# radius Called-Station, UAM '&called='
#	option mac_format ''		# MAC format specifier: 'aabbccddeeff', 'aa-bb-cc-dd-ee-ff', 'aa:bb:cc:dd:ee:ff' or the equivalent uppercase
#	option location_name ''		# radius WISPr-Location-Name

# for auth_mode 'uam':
#	option uam_port '3990'		# local UAM server port
#	option uam_secret ''		# remote UAM server password
#	option uam_server ''		# remote UAM server base url, e.g. "https://server.example.com/" - NB: trailing slash
#	option challenge ''		# UAM CHAP shared challenge
#	option final_redirect_url ''	# URL the client will be redirected to upon login. Special value 'uam' enables UAM 'success/reject/logoff' redirections URLs.
#	option mac_auth '0'		# Attempt MAC-authentication first
#	option mac_password ''		# Password sent for MAC-auth, defaults to MAC address
#	option mac_suffix ''		# Optional suffix appended to username for MAC-auth
#	option uam_sslurl ''		# optional base url to local UAM SSL (requires valid SSL setup in uhttpd UAM config), e.g. "https://uspot.lan:3991/" - NB: trailing slash
 ``` 
uhttpd:
 ``` 
config uhttpd 'main'
	list listen_https '0.0.0.0:443'
	list listen_https '[::]:443'
	option redirect_https '0'
	option home '/www'
	option rfc1918_filter '1'
	option max_requests '3'
	option max_connections '100'
	option cert '/etc/uhttpd.crt'
	option key '/etc/uhttpd.key'
	option cgi_prefix '/cgi-bin'
	list lua_prefix '/cgi-bin/luci=/usr/lib/lua/luci/sgi/uhttpd.lua'
	option script_timeout '60'
	option network_timeout '30'
	option http_keepalive '20'
	option tcp_keepalive '1'
	option ubus_prefix '/ubus'
[... 103 lines cut ...]
	option src_ip '192.168.205.161'	# adjust as needed
	option dest_port '3799'		# match value for 'das_port' in config/uspot
	option target 'ACCEPT'

# create the ipset that will hold authenticated clients
config ipset
	option name 'uspot'	# match with uspot option 'setname'
	list match 'src_mac'

# optional whitelist for e.g. remote UAM host and/or dynamic hosts via dnsmasq ipset functionality
config rule
	option name 'Allow-Whitelist'
	option src 'captive'
	option dest 'wan'
	option proto 'any'
	option ipset 'wlist'
	option target 'ACCEPT'

# associated whitelist ipset with prepopulated entries
config ipset
	option name 'wlist'
	list match 'dest_ip'
	list entry '192.168.205.161'	# adjust as needed for e.g. remote UAM server
#	list entry 'XX.XX.XX.XX'
 ```

AMArefkhani, 2024-02-17:
I added following attribute to the mentioned dictionary file in the previous comment, by considering the radius-client.c. 
radius-client.c
 ```
#define VENDORSPEC_WBAL			14122
#define ATTR_WBAL_WISPR_LOCATION_NAME	2
#define ATTR_WBAL_WISPR_LOGOFF_URL	3

#define VENDORSPEC_CHILLI		14559
#define ATTR_CHILLI_CHILLISPOT_LANG	7
 ```
dictionary:
 ```
.
.
.
ATTRIBUTE       WBAL-WISPR-Logoff-Url   60653528154115  string
ATTRIBUTE       WBAL-WISPR-Location-Name        60653528154114  string
ATTRIBUTE       ATTR-Chilli-Chillispot-Lang     62530428862471  string
.
.
.
```
But when I use radius-client it returns error "failed to lookup attribute key 22".

commit 53b8cb88 (2024-02-14): Makefile: require minimum ucode version

Fixes: #4