f00b4r0/uspot #9: Traffic accounting

AMArefkhani, 2024-04-04:
Hello, I need traffic accounting for my project. Could you please inform me, how long does it take to add the accounting traffic section?
I'm grateful for good support.

PWJW, 2024-04-15:
I wanted to also ask this.

Looking at the source code it looks like a lot of the work has already been done to support accounting interim/stop packets, so hopefully it isn't too much hassle to add this data to the `Acct-Input-Octets` and `Acct-Output-Octets` attributes in?

This is the number one reason why I can't switch over from coovachilli today, else I would be adopting uspot straight away as I love it.

PWJW, 2024-04-16:
@f00b4r0 Two related questions on this...

Looking at the code, I can see support for bandwidth limiting via `WISPr-Bandwidth-Max-Down` and `WISPr-Bandwidth-Max-Up`.

I have tested this (confirmed my RADIUS replied with these attributes) but the client gets full speed. Are there some other prerequisites for this to work?

 Also, the code indicates data usage limits can be set via `ChilliSpot-Max-Total-Octets` - is these implemented considering there is no accounting traffic monitoring at present?

Thanks

PWJW, 2025-03-17:
Hi just wondering if you have any update on this?

Thanks,

James

PWJW, 2025-03-26:
Fantastic, can I help in any way? Have you madd a start I can contribute towards?

f00b4r0, 2025-03-28:
I'll update this issue when there is something ready for testing. Please be patient.

f00b4r0, 2025-06-12:
It's [happening](https://github.com/f00b4r0/uspot/tree/ebpf), but not ready for testing yet. A little more patience and I will ask for testing feedback here. It's a matter of weeks at most now.

PWJW, 2025-06-12:
Fantastic. Ready to test as soon as you have a build ready 👍

badnbusy, 2025-06-14:
This is great - thanks - I will also test this out. Do you happen to have an openwrt compatible makefile for the new rbpf branch?

The current openwrt bundled one is: https://gitli.stratum0.org/ffbs/openwrt-packages/-/blob/ca2ff667e28e03d1b24a92696449f77b63e6dbf2/net/uspot/Makefile

f00b4r0, 2025-06-14:
The Makefile has been updated in the ebpf branch, you can use that. There are many ways to build this repo from source. Probably the quickest is to just delete the packages/net/uspot folder in your build tree and replace it with the entire content of the ebpf branch, that should build as you'd expect.

badnbusy, 2025-06-14:
Thanks, can they not co-exist? I thought traffic accounting wouldn't affect bandwidth (speed) throttling?

PWJW, 2025-06-15:
I've been testing this for around 8 hours thus far and all is working well. I can see the output in `uspot clients` with the additional traffic lines, plus I am receiving accounting packets with the traffic attributes to match.

Great work @f00b4r0.

Do you know if/when it will be possible to set a usage cap via the RADIUS Access-Accept reply so it disconnects the user when they reach the allowed usage limit returned?

Thanks  again, I will continue testing.

PWJW, 2025-06-18:
> [@PWJW](https://github.com/PWJW) For the record, can you confirm which hardware platform you tested on? Thanks

This was on an OpenWRT device, Cudy WR3000 (mediatek/filogic) on the latest OpenWRT snapshot.

commit 67ab9b30 (2025-06-13): uspot: hook traffic accounting

Fixes: #9
commit 468b9313 (2025-06-20): uspot: hook traffic accounting

Fixes: #9
commit 9bb9e24f (2025-06-20): uspot: hook traffic accounting

Fixes: #9
commit 319e0c28 (2025-07-02): uspot: hook traffic accounting

Fixes: #9