f00b4r0/uspot #11: Whitelisted domains / walled garden not working

PWJW, 2024-04-15:
I'm following the documentation in order to allow certain domains to be allowed before authentication (walled garden), but no matter what I put in the list, I cannot access these sites.

Am I missing something, or some other script? How does it match against the domains to allow traffic through the firewall (with them not being IP's)...

**/etc/config/firewall**

```
# optional whitelist for e.g. remote UAM host and/or dynamic hosts via dnsmasq ipset functionality
config rule
        option name 'Allow-Whitelist'
        option src 'captive'
        option dest 'wan'
        option proto 'any'
        option ipset 'wlist'
        option target 'ACCEPT'

# associated whitelist ipset with prepopulated entries
config ipset
        option name 'wlist'
        list match 'dest_ip'
        list entry '35.201.xx.xx' # my external portal IP - I can access this fine pre-auth
```

**/etc/config/dhcp**

```
config dhcp 'captive'
        option interface 'captive'
        option start '2'
        option limit '100'
        option leasetime '2h'

# if using optional dynamic hosts whitelist
config ipset
        list name 'wlist'       # match value with whitelist ipset name in config/firewall
        list domain 'sky.com'
        list domain 'bbc.co.uk'
        list domain 'itv.com'
```

Connecting to the SSID, opening a browser and trying to visit any of those 3 domains in the whitelist do not work ad it just gives me the usual cannot connect error like trying to visit any non-whitelisted domain.

Meanwhile. I can ccess the 35.201 IP I've set in the `config ipset` rule in the firewall.

Thoughts?

Thanks.

PWJW, 2024-04-15:
Hmm, it looks like "ipset" is not installed on my openwrt (custom build).

**$ ipset**
``
ipset: not found
```

and dnsmasq was built without ipset/nfset support:

**dnsmasq -v**

```
dnsmasq -v
Dnsmasq version 2.90  Copyright (c) 2000-2024 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile

```

What openwrt package(s) should be added to support ipset and domain whitelists, whist retaining default openwrt support for firewall4 and nft?

Thanks

pierrejoye, 2025-01-23:
Hello :)

Not sure if I can use this issue but I am stucked with the same issue. Everything works smoothly (need some adjustments from the README), but the white list.

As mentioned by @f00b4r0  (wishing I read this earlier), I tried as per the readme and no matter what, the domain in the dhcp config, or IP in the firewall, or both, that IP/address is not accessible.

After figuring out that dnsmasq does not have ipset but nftset, I tried the following:

dhcp:
```
config nftset                             
        option name 'wlist'               
        list match 'dest_ip'              
        list domain 'my.example.com'
```

firewall:
```
config rule                                                                          
        option name 'Allow-Whitelist'                                                
        option src 'guest'                                                           
        option dest 'wan'                                                            
        option proto 'any'                                                           
        option nftset 'wlist'                                                        
        option target 'ACCEPT'                                                       
        option log 1                                                                 
                                                                                     
config nftset                                                                        
        option name 'wlist'                                                          
        list match 'dest_ip'                                                         
        list entry '13.250.1.45'  
```

logread:
```
 daemon.info dnsmasq[1]: using standard nameservers for my.example.com
```

I also installed dnsmasq-full

I am surely missing something here. A dependency? or something else?

f00b4r0, 2025-01-23:
> Hello :)
> 
> Not sure if I can use this issue but I am stucked with the same issue. Everything works smoothly (need some adjustments from the README), but the white list.
> 
> As mentioned by [@f00b4r0](https://github.com/f00b4r0) (wishing I read this earlier), I tried as per the readme

Have you really? Your configuration is invalid.

> and no matter what, the domain in the dhcp config, or IP in the firewall, or both, that IP/address is not accessible.
>
> After figuring out that dnsmasq does not have ipset but nftset, I tried the [...]

section `nftset` does not exist, the correct section name is `ipset`; the option `name` is a `list`, not `option`, `list match 'dest_ip'` has nothing to do here.

https://openwrt.org/docs/guide-user/base-system/dhcp

> firewall:
>
> ```
> config rule
>         option name 'Allow-Whitelist' [...]

option `nftset` does not exist.
https://openwrt.org/docs/guide-user/firewall/firewall_configuration

> ```
> config [...]

section `nftset` does not exist.
https://openwrt.org/docs/guide-user/firewall/firewall_configuration

> logread:
> 
> ```
>  daemon.info dnsmasq[1]: using standard nameservers for my.example.com
> ```

I don't know what this is supposed to relate to but this has nothing to do with whitelisting.

> I also installed dnsmasq-full
> 
> I am surely missing something here. A dependency? or something else?

Using the correct configuration. If it doesn't work, please follow the [issue template](https://github.com/f00b4r0/uspot/blob/next/.github/ISSUE_TEMPLATE/bug_report.md) to provide sufficient information.

For what it's worth, the whitelisting is actively in use in large deployments of uspot, so it's pretty heavily tested.

pierrejoye, 2025-01-23:
> Have you really? Your configuration is invalid.

yes, my apologize if it was not clear. The config above was an attempt to get it work as the configuration using the README as base did not work .

> Using the correct configuration. If it doesn't work, please follow the [issue template](https://github.com/f00b4r0/uspot/blob/next/.github/ISSUE_TEMPLATE/bug_report.md) to provide sufficient [...]

> For what it's worth, the whitelisting is actively in use in large deployments of uspot, so it's pretty heavily tested.

I believe that and I love the quality of the code and simplicity of the project. That's the only part I am blocked with (I am no system admin or else for such things sadly). 

I will open using the template if I can't figure it out. Thanks :)