f00b4r0/uspot #13: MAC authentication still performing captive portal popup

PWJW, 2024-04-23:
We are testing the MAC authentication option within uspot and although it's working and the user is successfully authenticated from our RADIUS server, they still receive the captive portal popup window (which then briefly disappears on Android or the window changes to "Done" on iOS). 

I've followed the instructions to have dnsmasq handle this via the custom script and then auth the device using ubus but they still get the popup.

"hotspot" is the uspot name/interface
`$MACADDR` and `$IPADDR` comes through via the dhcp script.

```
ubus call uspot client_auth "{ \"uspot\":\"hotspot\", \"address\":\"$MACADDR\", \"client_ip\":\"$IPADDR\" }" > /dev/null
ubus call uspot client_enable "{ \"uspot\":\"hotspot\", \"address\":\"$MACADDR\" }" 2>/dev/null
fi
```

Is this expected behaviour or am I missing something? It's like we need the mac authentication to happen before the client device tries to check for Internet connectivity and this is what redirects to the CPD code and shows the portal page...

Thanks

f00b4r0, 2024-04-23:
if you remove the `/dev/null` redirections after each call you'll see in the logs whether or not the clients are authenticated via this mechanism. It may happen that some clients are "too quick" in testing connectivity and do show a popup anyway, this is outside of uspot control (you'll notice the popups display device-generated messages without showing the uspot pages) and so not an uspot issue.

The DHCP-based auth bypass is more a hack than a proper feature anyway so your mileage may vary anyway. MAC-auth is supported by uspot via the web interface (without the bypass script), where a uspot-generated message would be displayed.

closed 2024-04-23 by f00b4r0

PWJW, 2024-04-23:
The auth was successful as the logs show it :)

MAC auth works "properly" with coovachilli and all other wifi vendors I've used like Cisco, Meraki, Ruckus, Aruba etc, so I don't think its an issue with the client. On all of these other types of implementations, the client never sees any popup or redirect and is immediately online when the RADIUS replies with Access-Accept (before the client tries to call a URL).

The traditional purpose of MAC auth is to authenticate a device before a captive portal is involved so I feel the current implementation isn't really usable as a true MAC auth solution in the current way it's implemented. It will work for some use cases though!

Is there any plan to revisit how this is implemented at all?

Thanks again.

f00b4r0, 2024-04-23:
> Is there any plan to revisit how this is implemented at all?

No, but this is opensource: patches are always welcome.