f00b4r0/uspot #14: Unable to dispaly login Form

BDZK, 2024-06-22:
Hi
I have installed your software on openwrt.
i have this error on login page Error, An error occured. Please retry.

in the log i have :
daemon.err uhttpd[4138]: uspot: 10.1.8.178 - config not found for "null"

Can you help me please?

BDZK, 2024-06-23:
It seems that the login script cannot get the mac address, IP etc. of the client. I wonder why. Maybe i nee to install manually a dependency?

f00b4r0, 2024-06-24:
Please provide:

- OpenWrt version
- output of `opkg list-installed | grep uspot`
- contents of `/etc/config/uspot`, `/etc/config/firewall`, `/etc/config/dhcp`, `/etc/config/uhttpd` - mask private information such as passwords
- output of `logread -e uspot` when the problem occurs

f00b4r0, 2024-06-24:
Please follow the instructions https://github.com/f00b4r0/uspot?tab=readme-ov-file#radcli

BDZK, 2024-06-25:
Now  if i submit the form i have 'invalid credentials'.

logread -e uspot is giving me this:
Tue Jun 25 02:59:09 2024 daemon.err uhttpd[2558]: uspot: 10.1.8.44 - start radius flow
Tue Jun 25 02:59:26 2024 daemon.err uhttpd[2558]: uspot: 10.1.8.234 - invalid credentials
Tue Jun 25 02:59:26 2024 daemon.err uhttpd[2558]: uspot: 10.1.8.234 - start radius flow
Tue Jun 25 02:59:26 2024 daemon.err uhttpd[2558]: uspot: 10.1.8.234 - start radius flow

It seems that the form is not even submitting.

If i inspect in the browser is that it is requestion a missing favicon????

f00b4r0, 2024-06-25:
Please provide **all** the requested data.

BDZK, 2024-06-25:
Here are the requested data :
- Openwrt 23.05.3
- logread -e uspot
Tue Jun 25 09:14:20 2024 daemon.debug uspot: captive acct-on call
Tue Jun 25 09:14:37 2024 daemon.err uhttpd[2590]: uspot: 10.1.8.44 - start credentials flow
Tue Jun 25 09:14:37 2024 daemon.err uhttpd[2590]: uspot: 10.1.8.44 - start credentials flow
Tue Jun 25 09:14:44 2024 daemon.err uhttpd[2590]: uspot: 10.1.8.44 - invalid credentials
Tue Jun 25 09:14:44 2024 daemon.err uhttpd[2590]: uspot: 10.1.8.44 - start credentials flow
Tue Jun 25 09:14:44 2024 daemon.err uhttpd[2590]: uspot: 10.1.8.44 - start credentials flow

Other files attached.

Thanks.
[dhcp.txt](https://github.com/user-attachments/files/15968330/dhcp.txt)
[firewall.txt](https://github.com/user-attachments/files/15968331/firewall.txt)
[httpd.txt](https://github.com/user-attachments/files/15968333/httpd.txt)
[uspot.txt](https://github.com/user-attachments/files/15968335/uspot.txt)

f00b4r0, 2024-06-25:
Option `nas_macˋ is unset, I think that's the problem.

BDZK, 2024-06-25:
Still not working... I have set it with the mac of the captive interface but still not working

f00b4r0, 2024-06-25:
You either have a configuration error in uspot or a configuration error in your RADIUS server. With no further information there is very little I can do. You can try setting `option debug '1'` in the uspot interface configuration to expand further error messages from uspot

f00b4r0, 2024-09-17:
Do you have further input to provide or has this issue been resolved for you?

f00b4r0, 2025-01-20:
no news after 6 months, closing.

closed 2025-01-20 by f00b4r0