f00b4r0/uspot #17: Whitelist domain or IP remains blocked

pierrejoye, 2025-01-23:
**Describe the bug**

As explained earlier, I am pretty there is something preventing this working feature to function correctly. I just can't see what.

Steps to reproduce the behavior:

Add a domain to the whitelist dhcp config, an IP to the firewall, or both. 

**Expected behavior**

IP or domain accessible while not authenticated.

**Configuration**
Provide the output of:

OpenWrt 23.05.2
```
Linux OpenWrt 5.15.137 #0 SMP Tue Nov 14 13:38:11 2023 mips GNU/Linux
```

```bash
opkg list-installed | grep uspot
uspot - 2024-05-03-92d3356d-1
uspot-www - 2024-05-03-92d3356d-1
uspotfilter - 2024-05-03-92d3356d-1

```

```
opkg list-installed |grep dns
dnsmasq-full - 2.90-2
rpcd-mod-rrdns - 20170710
```

```
 opkg list-installed |grep nft
kmod-nft-core - 5.15.137-1
kmod-nft-fib - 5.15.137-1
kmod-nft-nat - 5.15.137-1
kmod-nft-offload - 5.15.137-1
libnftnl11 - 1.2.6-1
nftables-json - 1.0.8-1
```

- `uci show uspot` (replace passwords with XXXX as needed)
- `uci show network`

```
network.loopback=interface
network.loopback.device='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.globals=globals
network.globals.ula_prefix='fdca:ad65:2a4c::/48'
network.globals.packet_steering='1'
network.@device[0]=device
network.@device[0].name='br-lan'
network.@device[0].type='bridge'
network.@device[0].ports='lan1' 'lan2' 'lan3' 'lan4'
network.lan=interface
network.lan.device='br-lan'
network.lan.proto='static'
network.lan.ipaddr='192.168.2.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.wan=interface
network.wan.device='wan'
network.wan.proto='dhcp'
network.wan.type='bridge'
[... 312 lines cut ...]
Thu Jan 23 10:43:05 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Thu Jan 23 10:43:05 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Thu Jan 23 10:43:05 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Thu Jan 23 10:43:05 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Thu Jan 23 10:43:05 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 6 names
Thu Jan 23 10:43:05 2025 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Thu Jan 23 10:43:05 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
```

log from the wifi connection session start,, click to continue page correctly display, before continuing I try to go the whitelisted IP and it fails
```
Thu Jan 23 10:43:41 2025 daemon.notice hostapd: phy0-ap2: AP-STA-DISCONNECTED da:37:55:33:53:bb
Thu Jan 23 10:43:41 2025 daemon.info hostapd: phy0-ap2: STA da:37:55:33:53:bb IEEE 802.11: disassociated
Thu Jan 23 10:43:42 2025 daemon.info hostapd: phy0-ap2: STA da:37:55:33:53:bb IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu Jan 23 10:43:47 2025 daemon.info hostapd: phy0-ap2: STA da:37:55:33:53:bb IEEE 802.11: authenticated
Thu Jan 23 10:43:47 2025 daemon.info hostapd: phy0-ap2: STA da:37:55:33:53:bb IEEE 802.11: associated (aid 1)
Thu Jan 23 10:43:47 2025 daemon.notice hostapd: phy0-ap2: AP-STA-CONNECTED da:37:55:33:53:bb auth_alg=open
Thu Jan 23 10:43:48 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-guest) 192.168.3.173 da:37:55:33:53:bb
Thu Jan 23 10:43:48 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-guest) 192.168.3.173 da:37:55:33:53:bb Pierre-s-S21-Ultra
Thu Jan 23 10:43:50 2025 daemon.err uhttpd[27459]: uspot: 192.168.3.173 - start click-to-continue flow
Thu Jan 23 10:43:50 2025 daemon.err uhttpd[27459]: uspot: 192.168.3.173 - start click-to-continue flow
Thu Jan 23 10:43:51 2025 daemon.err uhttpd[27459]: uspot: 192.168.3.173 - start click-to-continue flow
Thu Jan 23 10:43:51 2025 daemon.err uhttpd[27459]: uspot: 192.168.3.173 - start click-to-continue flow

```

f00b4r0, 2025-01-23:
Hi,

If you expect me to spend time trying to fix your problem, please have the courtesy of providing *ALL* the information I need.

>  * `uci show uspot` (replace passwords with XXXX as needed)

Please provide this. You can remove all password settings there are unrelated here.

>  * `uci show firewall`

Please provide the output of that command instead of the file. This will show what UCI actually parsed, not what you typed (which may contain typos which are difficult to spot).

> * `uci show dhcp`

Please provide this, it's essential to diagnose this problem.

pierrejoye, 2025-01-23:
> Hi,
> 
> If you expect me to spend time trying to fix your problem, please have the courtesy of providing _ALL_ the information I need.

Sorry, I am getting tired :)  Markup mistakes hide the uspot's one.

> Please provide this. You can remove all password settings there are unrelated here.
>
>
> Please provide the output of that command instead of the file. This will show what UCI actually parsed, not what [...]

Edited the report, replaced with the uci show and fixed the mark up.

f00b4r0, 2025-01-23:
>  * `uci show firewall`
> 
> ```
> firewall.@defaults[0]=defaults
> firewall.@defaults[0].syn_flood='1'
> firewall.@defaults[0].input='REJECT'
> firewall.@defaults[0].output='ACCEPT'
> firewall.@defaults[0].forward='REJECT'
> firewall.@zone[0]=zone
> firewall.@zone[0].name='lan'
> firewall.@zone[0].network='lan'
> firewall.@zone[0].input='ACCEPT'
> firewall.@zone[0].output='ACCEPT'
> firewall.@zone[0].forward='ACCEPT'
> firewall.@zone[1]=zone
> firewall.@zone[1].name='wan'
> firewall.@zone[1].network='wan' 'wan6'
> firewall.@zone[1].input='REJECT'
> firewall.@zone[1].output='ACCEPT'
> firewall.@zone[1].forward='REJECT'
> firewall.@zone[1].masq='1'
> firewall.@zone[1].mtu_fix='1'
> firewall.@forwarding[0]=forwarding
> firewall.@forwarding[0].src='lan'
> firewall.@forwarding[0].dest='wan'
> firewall.@rule[0]=rule
> firewall.@rule[0].name='Allow-DHCP-Renew'
> firewall.@rule[0].src='wan'
> firewall.@rule[0].proto='udp'
> firewall.@rule[0].dest_port='68'
> firewall.@rule[0].target='ACCEPT'
> firewall.@rule[0].family='ipv4'
> firewall.@rule[1]=rule
> firewall.@rule[1].name='Allow-Ping'
> firewall.@rule[1].src='wan'
> firewall.@rule[1].proto='icmp'
> firewall.@rule[1].icmp_type='echo-request'
> firewall.@rule[1].family='ipv4'
> firewall.@rule[1].target='ACCEPT'
> firewall.@rule[2]=rule
> firewall.@rule[2].name='Allow-IGMP'
> firewall.@rule[2].src='wan'
> firewall.@rule[2].proto='igmp'
> firewall.@rule[2].family='ipv4'
> firewall.@rule[2].target='ACCEPT'
> firewall.@rule[3]=rule
> firewall.@rule[3].name='Allow-DHCPv6'
> firewall.@rule[3].src='wan'
> firewall.@rule[3].proto='udp'
> firewall.@rule[3].dest_port='546'
> firewall.@rule[3].family='ipv6'
> firewall.@rule[3].target='ACCEPT'
> firewall.@rule[4]=rule
> firewall.@rule[4].name='Allow-MLD'
> firewall.@rule[4].src='wan'
> firewall.@rule[4].proto='icmp'
> firewall.@rule[4].src_ip='fe80::/10'
> firewall.@rule[4].icmp_type='130/0' '131/0' '132/0' '143/0'
> firewall.@rule[4].family='ipv6'
> firewall.@rule[4].target='ACCEPT'
> firewall.@rule[5]=rule
> firewall.@rule[5].name='Allow-ICMPv6-Input'
> firewall.@rule[5].src='wan'
> firewall.@rule[5].proto='icmp'
> firewall.@rule[5].icmp_type='echo-request' 'echo-reply' 'destination-unreachable' 'packet-too-big' 'time-exceeded' 'bad-header' 'unknown-header-type' 'router-solicitation' 'neighbour-solicitation' 'router-advertisement' 'neighbour-advertisement'
> firewall.@rule[5].limit='1000/sec'
> firewall.@rule[5].family='ipv6'
> firewall.@rule[5].target='ACCEPT'
> firewall.@rule[6]=rule
> firewall.@rule[6].name='Allow-ICMPv6-Forward'
> firewall.@rule[6].src='wan'
> firewall.@rule[6].dest='*'
> firewall.@rule[6].proto='icmp'
> firewall.@rule[6].icmp_type='echo-request' 'echo-reply' 'destination-unreachable' 'packet-too-big' 'time-exceeded' 'bad-header' 'unknown-header-type'
> firewall.@rule[6].limit='1000/sec'
> firewall.@rule[6].family='ipv6'
> firewall.@rule[6].target='ACCEPT'
> firewall.@rule[7]=rule
> firewall.@rule[7].name='Allow-IPSec-ESP'
> firewall.@rule[7].src='wan'
> firewall.@rule[7].dest='lan'
> firewall.@rule[7].proto='esp'
> firewall.@rule[7].target='ACCEPT'
> firewall.@rule[8]=rule
> firewall.@rule[8].name='Allow-ISAKMP'
> firewall.@rule[8].src='wan'
> firewall.@rule[8].dest='lan'
> firewall.@rule[8].dest_port='500'
> firewall.@rule[8].proto='udp'
> firewall.@rule[8].target='ACCEPT'
> firewall.@zone[2]=zone
> firewall.@zone[2].name='guest'
> firewall.@zone[2].network='guest'
> firewall.@zone[2].input='REJECT'
> firewall.@zone[2].output='ACCEPT'
> firewall.@zone[2].forward='REJECT'
> firewall.@redirect[0]=redirect
> firewall.@redirect[0].name='Redirect-unauth-captive-CPD'
> firewall.@redirect[0].src='guest'
> firewall.@redirect[0].src_dport='80'
> firewall.@redirect[0].proto='tcp'
> firewall.@redirect[0].target='DNAT'
> firewall.@redirect[0].reflection='0'
> firewall.@redirect[0].ipset='!uspot'
> firewall.@rule[9]=rule
> firewall.@rule[9].name='Allow-DHCP-NTP-captive'
> firewall.@rule[9].src='guest'
> firewall.@rule[9].proto='udp'
> firewall.@rule[9].dest_port='67 123'
> firewall.@rule[9].target='ACCEPT'
> firewall.@rule[10]=rule
> firewall.@rule[10].name='Restrict-input-captive'
> firewall.@rule[10].src='guest'
> firewall.@rule[10].dest_ip='!guest'
> firewall.@rule[10].target='DROP'
> firewall.@rule[11]=rule
> firewall.@rule[11].name='Allow-captive-CPD-WEB-UAM'
> firewall.@rule[11].src='guest'
> firewall.@rule[11].dest_port='80 443 3990'
> firewall.@rule[11].proto='tcp'
> firewall.@rule[11].target='ACCEPT'
> firewall.@rule[12]=rule
> firewall.@rule[12].name='Forward-auth-captive'
> firewall.@rule[12].src='guest'
> firewall.@rule[12].dest='wan'
> firewall.@rule[12].proto='any'
> firewall.@rule[12].target='ACCEPT'
> firewall.@rule[12].ipset='uspot'
> firewall.@rule[13]=rule
> firewall.@rule[13].name='Block-unauth-guest-internet'
> firewall.@rule[13].src='guest'
> firewall.@rule[13].dest='wan'
> firewall.@rule[13].proto='any'
> firewall.@rule[13].target='DROP'
> firewall.@rule[13].ipset='!uspot'
> firewall.@rule[14]=rule
> firewall.@rule[14].name='Allow-DNS-captive'
> firewall.@rule[14].src='guest'
> firewall.@rule[14].proto='udp' 'tcp'
> firewall.@rule[14].dest_port='53'
> firewall.@rule[14].target='ACCEPT'
> firewall.@ipset[0]=ipset
> firewall.@ipset[0].name='uspot'
> firewall.@ipset[0].match='src_mac'
> firewall.@rule[15]=rule
> firewall.@rule[15].name='Allow-Whitelist'
> firewall.@rule[15].src='guest'
> firewall.@rule[15].dest='wan'
> firewall.@rule[15].proto='any'
> firewall.@rule[15].ipset='wlist'
> firewall.@rule[15].target='ACCEPT'
> firewall.@ipset[1]=ipset
> firewall.@ipset[1].name='wlist'
> firewall.@ipset[1].match='dest_ip'
> firewall.@ipset[1].entry='13.234.12.223'
> ```

Rule 13, which you added and which is not part of the documentation, unconditionally blocks all unauthenticated traffic. It is ordred before the whitelist rule, Rule 15, which is thus never evaluated.

This is a firewall misconfiguration.

closed 2025-01-23 by f00b4r0

pierrejoye, 2025-01-23:
Thanks! Now it works perfectly. I think it was a remaining from opennds config I used as base for the setup scripts.

Side question, are you open for a small PR for the README and/or documentation? Easy ones about install from a fresh flashed openwrt? like which packages, uspot uspotfilter and usport-www (I suppose only used for local not needed for UAM, need to check).  I will do some scripts to test the flow for the routers we use here.

f00b4r0, 2025-01-23:
> Side question, are you open for a small PR for the README and/or documentation? Easy ones about install from a fresh flashed openwrt? like which packages, uspot uspotfilter and usport-www (I suppose only used for local not needed for UAM, need to check).

Normally everything should already be covered in README and package dependencies, but feel free to submit documentation patches.