f00b4r0/uspot #21: final_redirect_url not working

avseeval, 2025-02-13:
Hello. I installed and configure uspot, everything works, but option final_redirect_url not working, after authorisation I get only white page with word Success

[image]

[network.txt](https://github.com/user-attachments/files/18788833/network.txt)
[uspot.txt](https://github.com/user-attachments/files/18788831/uspot.txt)
[firewall.txt](https://github.com/user-attachments/files/18788832/firewall.txt)
[dhcp.txt](https://github.com/user-attachments/files/18788834/dhcp.txt)

f00b4r0, 2025-02-16:
Hi, I cannot reproduce your issue. The screen you captured (from `captive.apple.com`) is usually shown by iOS devices when the captive portal detection was "successful", i.e. the device was already authenticated and the UAM portal was not involved. I see from dhcp.txt that you use the DHCP MAC auth "hack": this bypasses the captive portal and can definitely result in what you're seeing here.

avseeval, 2025-02-17:
Hi, when the captive portal was detected - I see uam portal server. Then, from server I got logon url with username and passwod, but after succes authentication I get this, but not redirect to final_redirect_url. Yes, I use  DHCP MAC auth "hack", but not always it bypasses the captive portal. I think that in many cases firewall and uhttpd work much faster than DHCP MAC auth "hack", so the captive window still appears, but it immediately shows success. Also, using debug, I was able to find that the mac_passwd option does not make sense, since the mac is also transmitted as a password

f00b4r0, 2025-02-17:
If the captive.apple.com window is shown then the portal (uhttpd) was never involved. This is almost certainly due to DHCP MAC auth "hack", so if you still think there is a bug, please disable that and try again. As mentioned in the doc the DHCP auth is an unsupported "hack". The `mac_passwd` option is only used for MAC authentication (option `mac_auth` enabled) where the provided password replaces the formatted MAC for the RADIUS `Call-Check`, see:
https://github.com/f00b4r0/uspot/blob/644fd6f52a8e7b1a72e0937ad0d4cea8e86186a9/files/usr/share/uspot/uspot.uc#L744