f00b4r0/uspot #43: Option to specify interface for getting NAS-IP-Address in accounting packets

pandafy, 2025-09-03:
Currently, it is not possible to set the NAS IP in uspot. It would be very useful to have an option to specify the interface from which the NAS-IP-Address is set in accounting packets.

**Background:**
The NAS-IP-Address attribute can be used for CoA (Change of Authorization) requests, and therefore it may not always coincide with the IP address of the WAN interface. For example, in my testing, the NAS IP was set to a local address (e.g., `192.168.x.x`), which is not reachable for CoA requests. In @openwisp, the desired NAS IP is actually the IP of a VPN interface. 

**Feature Request:**
Add an option to specify the interface from which to get the NAS-IP-Address when generating accounting packets. This would provide much more flexibility for deployments where the management interface differs from the main network interface (wan).

nemesifier, 2025-09-04:
This need arises as a consequence of using RadSec to communicate with a RADIUS server using a public IPv4.

nemesifier, 2025-09-05:
@f00b4r0 manually specifying an IP is ok.

closed 2025-09-30 by f00b4r0